marcel-licence/esp32_midi_sampler #8: Same note whatever key is pressed - SD card

Slider2732, 2021-06-22:
First of all - fantastic project !
I bought 2x the Audio Kit V2.2 and they arrived today. ESP32-CAM was selected, couldn't find the Audio Kit in the Arduino IDE boards section. It seems to work fine, displaying PSRAM as above 0 bytes in the IDE terminal and enabling the code to run.

Problem is, even though a sample is loaded from the SD card at switch on, it plays the same note on every key. Polyphony seems to be there, except the same note is played several times at once. 

My keyboard is a MIDI'd 37 key PAiA keybed, using an Arduino. Output channel is channel 1. It works great with your Easy Synth and the Supersaw version of it (2 separate circuits have been built).
In your video, it seems channel 1 is used differently to the other channels with the sampler?
I've looked through the MIDI section but am having trouble finding the part where channel 1 may be allowed to work like the other channels... presumably enabling notes to be transposed up and down. 

Other issues for another time would be implementation of the multiplexer sliders/knobs code to work on this project too and to output the VT100 type data to a small screen (ST7735). I'd like to select samples from the SD card in the same way as you show in the video  :)   

Thanks for any help. The sound of the loaded Korg M1 piano sample is high quality on this great project of yours, I just need it to change pitch !

marcel-licence, 2021-06-23:
Hey,
yes channel 1 is to play different samples in the original recorded key. You should move over to another channel to play another sample.
You could manipulate Sampler_NoteoOn and force the channel value to 1 for example to be able playing one sample at different pitches.
I also started to adopt a display. It is .... yeah not so easy on the audio kit xD
I've to watch my video again .. so long time ago ;-)

Slider2732, 2021-06-23:
Thanks for the reply  :)
Whole idea at the moment is to get it functional with basic changes, like next sample loading. The project deserves to be finished with CC commands (they work very well in your video), just my MIDI controller is USB not a full sized connector. 

There is a related thread where someone has cut resistor lines for an SD1306, which may be the route? Luckily the boards are large compared to regular ESP32. I2C 2x signal line displays would seem to suit the Audio Kit.  

Have altered the code for button press selection of the next sample on the SD card, It compiles, but doesn't work. The code does say about the wrong area for the buttons to be (volume up/down and play). That's where I've put the changes and run into problems when removing '#if 0 /* wrong place */'. SD_MMC examples are more difficult to find than sd.h too. Am not even sure whether it's reading the SD card file names after booting and autoloading the first one. 
At the moment, I'm changing the filenames of each sample to 'M1_Piano.wav' and replacing on the SD card, which isn't ideal!
Mind you, it's a lot of fun as is. Have wished for such hardware since about 1990.

marcel-licence, 2021-06-24:
Yes the stuff is a bit late on the market. Ihe things were too expensive years ago or not powerful enough.
Yes there are some capacitors there causing the trouble. I try to avoid using the buttons because you are losing GPIOs for other functions.
Unfortunately it will cost me some time to bring all project to the same state because I tried to establish a clean structure to make it easier to adjust them or replace parts.

Following will preload a file from littleFs:
```
#ifdef AUTO_LOAD_PATCHES_FROM_LITTLEFS
    /* finally we can preload some data if available */
    PatchManager_SetDestination(0, 1);
    Sampler_LoadPatchFile("/samples/pet_bottle.wav");
#endif
```

Slider2732, 2021-06-24:
Thanks. 
The above section of code was changed similarly for the M1 piano sound on the SD card. That's how I can play different samples, by renaming them each time to the same name  :)
I'd like to call the file load function in the same way as you show from your Edirol. Something similar to:

If (PIN_KEY_1 == LOW)
  {
   File root = fs.open(samples);
   File file = root.openNextFile();
   strcopy(file, file);
   Sampler_LoadPatchFile(file);
 }

But obviously not that lol
I think it's this function I need to call, because you've already written the routines.
uint32_t PatchManager_LoadWavefile(fs::FS &fs, char *filename, int16_t *buffer, uint32_t bufferSize)

Seems like it could be done with a 'fake' CC message(?). Such as pressing a button and it sends the correct CC to the patch manager, as though a MIDI controller sent it. 
But yes, it would likely be temporary, with screen support being worked on and only a few spare pins.  

The clean structure is a good idea. A set of tools being built, slotting in as needed. 
Any further work on these projects is always appreciated and I won't be alone in saying that!

Slider2732, 2021-06-25:
Ooops.. I unintentionally Forked your code to my channel!
Am new here and clicked the Fork button, wanting to see who or what had changed in someone elses version. Anyway, am not intending to make changes as a Fork, am looking for how to delete it.

Did try resoldering a cheap DIN MIDI to USB cable and box on my Worlde MIDI controller, such that your code could be used correctly and fully. But the cable only goes from DIN to USB, not the other way around. 

A new sample loaded from SD card still needs to have the samples storage reset for each load. Still workin' on it   :)

marcel-licence, 2021-06-26:
No problem, in general fork is the best way to get a copy and modify it =) So you have also the tracking included.
You are right!!! I should try to resolder the keys. Not all components are placed that looks like you can modify it. Very good point. I've seen IO36 but never thought about using it xD 

Hm your samples are incomplete? Hm maybe the format is not supported. (s16le, mono, 44100samples/s)

marcel-licence, 2021-06-26:
You can clear/delete a sample by noteOn/off of the sample you want to delete followed by Sampler_RemoveActiveRecording(0,1)

Slider2732, 2021-06-26:
Many thanks Marcel, it works!
Only change needed was to add the line to my code above:

  Sampler_RemoveActiveRecording(0,1);
  PatchManager_SetDestination(1, 1);
  Sampler_LoadPatchFile(Next);

It now steps through all 12 of the numbered patches. Next idea is to have a long press skip by 10 patches. Then it would make storing any number of patches easier to manage. It takes until the next Blink toggle to work, about 1/2 second typically, so I press a low note then Key1 and wait for the audio interference while the next patch is loaded... might change that to an audible beep as confirmation.
 
The wider thinking is to put features on the buttons for people who may only have an interest in music on microcontrollers, they'd have no controller and yet want to explore your work. Features would be limited, but somewhat usable. Long press Key2 for sample source select, short press to begin the auto level sampling. Key3 for basic looping (edit loop points in Audacity maybe to set them up). Key4 add a set % reverb. Etc.
Maybe I could use the fork to do that  :)  

Only trouble is some loss of notes, which I think is from the call to digitalRead IO36 in the Blink section. Sometimes chords work (4 notes), sometimes only 1 note will play.
All samples are 44100, 16 bit wav, found online. 

Btw, with supersaw, Easy Synth and now the sampler, I call my keyboard build The Licence'd Synth. A play on your name, when pronounced like driving licence.
Here it is, before the sampler addition: 
https://ibb.co/nbBr9yH

closed 2021-06-30 by Slider2732