marcel-licence/esp32_midi_sampler #55: ES8388 volume

laurence-diack, 2021-09-29:
Just a question regarding the ES8388 driver implementation, do you find that the output volume is significantly lower using the onboard codec vs an external dac like 5102 or 1334a? I've tried changing a few settings but it seems all the relevant output / gain levels are set to maximum in your setup() method anyway.

Anyway, thank you for the excellent project!

marcel-licence, 2021-09-30:
Do you use line out with an additional amplifier or do you try to use headphones?
There is some float to int conversation which could cause a loss of max volume:

```
#ifdef SAMPLE_SIZE_16BIT
        sampleDataU[n].ch[0] = int16_t(fr_sample[n] * 16383.0f); /* some bits missing here */
        sampleDataU[n].ch[1] = int16_t(fl_sample[n] * 16383.0f);
#endif
#ifdef SAMPLE_SIZE_32BIT
        sampleDataU[n].ch[0] = int32_t(fr_sample[n] * 1073741823.0f); /* some bits missing here */
        sampleDataU[n].ch[1] = int32_t(fl_sample[n] * 1073741823.0f);
#endif
```

laurence-diack, 2021-09-30:
Thanks for the suggestions, I was using headphones but line in to mixer works a little better. I guess the ES8388 is just not a particularly hot output. I will experiment with different multiplier in the float to int as well since nothing I was putting through it seemed to get close +/-1.0f anyway.

marcel-licence, 2021-10-02:
It seems that it is not recommended to use headphones. There could be a way. But actually other users detected that especially the low frequencies are cut away.

closed 2021-10-02 by marcel-licence