marcel-licence/esp32_midi_sampler #68: How to play samples from buttons or custom Sensors ?

baesek, 2022-08-25:
Hi, I have a ESP32 Audio Kit V2.2 board with ES8388 Codec and i just want to make it play a sound through the headphone output when i push one of the 6 buttons. To do so I implemented basic button reads by adding 
#define AUDIO_KIT_BUTTON_DIGITAL
to the config.h instead of the #define AUDIO_KIT_BUTTON_ANALOG
Then i implemented it as follows in the esp32_audio_kit_module.h file. I receive the button events up and down, but I cannot really play the samples i uploaded the data folders contents to the psram using the mylittlefs plugin for the arduino ide already.
I tried to get it act like this: https://www.youtube.com/watch?v=r0af0DB1R68
`
In the end i want to read sensors and playback samples as they reach a treshold... 
...

#ifdef AUDIO_KIT_BUTTON_DIGITAL
/*
 * when not modified and R66-R70 are placed on the board
 */

#define PIN_KEY_1                   (36)
#define PIN_KEY_2                   (13)
#define PIN_KEY_3                   (19)
#define PIN_KEY_4                   (23)
#define PIN_KEY_5                   (18)
#define PIN_KEY_6                   (5)
/* My edit */
uint8_t buttonMap[6] = { PIN_KEY_1, PIN_KEY_2, PIN_KEY_3, PIN_KEY_4, PIN_KEY_5, PIN_KEY_6 };
uint8_t buttonStates[sizeof(buttonMap)];
/* End of edit */ 

...

typedef void(*audioKitButtonCb)(uint8_t, uint8_t);
extern audioKitButtonCb audioKitButtonCallback;

...

/*
 * pullup required to enable reading the buttons (buttons will connect them to ground if pressed)
 */
void button_setup()
{
#ifdef AUDIO_KIT_BUTTON_DIGITAL
    // Configure keys on ESP32 Audio Kit board
/* My edit */
    for (uint8_t i = 0; i < sizeof(buttonMap); i++) {
      pinMode(buttonMap[i], INPUT_PULLUP);
    }
/* End of My Edit */
#endif
...
}
...

/*
 * very bad implementation checking the button state
 * there is some work required for a better functionality
 */
void button_loop()
{
#ifdef AUDIO_KIT_BUTTON_DIGITAL
    /* My edit */
    static uint8_t buttonCount = 6;//sizeof(buttonMap);
    static bool state;
    
    for (uint8_t i = 0; i < buttonCount; i++) {
      state = digitalRead(buttonMap[i]);
      if (state != buttonStates[i]) {
        if (state == LOW) {
          Serial.println("Button "); Serial.print(i+1 ); Serial.println(" down");
          if (audioKitButtonCallback != NULL)
          {
              audioKitButtonCallback(i, 1);
          }
        } else {
          Serial.println("Button "); Serial.print(i+1); Serial.println(" up");
          if (audioKitButtonCallback != NULL)
          {
              audioKitButtonCallback(i, 0);
          }
        }
        buttonStates[i] = state;
      }
    }
    /* End of my edit */
#endif
...
}

#endif`

marcel-licence, 2022-08-27:
Hi,
does the firmware work without any modifications except setting up I2S and I2C to ensure that the codec is working?
It might be possible that there are some GPIO conflicts. At the moment I do not have any idea where the error from the Wire library came from.
Best regards,
Marcel

baesek, 2022-08-29:
Hi Marcel, 
Thank you very much for your answer(time), I just cloned the repo again and I just uncommented 
#define NOTE_ON_AFTER_SETUP
After upload I get no sound and this on the arduino Monitor(i set verbose output at compilation and upload in the arduino settings) :
```
⸮⸮ ⸮ ⸮⸮⸮ ⸮   ⸮ ⸮⸮⸮⸮⸮  A⸮zA⸮⸮⸮⸮..⸮⸮⸮⸮0,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[ ⸮⸮mum⸮⸮⸮⸮2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   451][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

Connect to ES8388 codec... 0x00: 0x05
[  1071][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
0x01: 0x40
[  1084][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[  1087][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[  1099][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
AdcCh1!
MixChAMPL!
[  1117][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[  1126][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
ES8388 setup finished!
Reg 0x00 = 0x05
[  1149][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x01 = 0x40
[  1165][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x02 = 0x00
[  1179][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x03 = 0x00
[  1192][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x04 = 0x3c
[  1206][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x05 = 0x00
[  1220][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x06 = 0x00
[  1234][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x07 = 0x7c
[  1248][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x08 = 0x00
[  1262][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x09 = 0x88
[  1276][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x0a = 0x50
[  1290][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x0b = 0x02
[  1304][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x0c = 0x0c
[  1317][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x0d = 0x02
[  1331][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x0e = 0x30
[  1345][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x0f = 0x20
[  1359][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x10 = 0x00
[  1373][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x11 = 0x00
[  1387][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x12 = 0x16
[  1401][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x13 = 0xb0
[  1415][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x14 = 0x32
[  1429][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x15 = 0x06
[  1442][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x16 = 0x00
[... 83 lines cut ...]
Free heap: 143464
Total PSRAM: 4192059
Free PSRAM: 18679
Firmware started successfully
[?25l[0;0HLoaded sample: 22049            
--------------------------------
inL    [0;32m        [0m  outL   [0;32m####    [0m
inR    [0;32m        [0m  outR   [0;32m####    [0m
--------------------------------
56926 of 2064378 bytes used     
O_______________________________
[... cut ...]

```
What do you mean be setting up I2S and I2C? I couldn't find any relating setting in the config.h nor in the [README.md](https://github.com/marcel-licence/esp32_midi_sampler#readme)

I dont know if this is relevant, but:
The Key2 on my board is only working if set Dip Switches 1, 2, 3 to ON and 4,5 to OFF, If I set only 2 and 3 to ON it is not working. (detectable by my test code)

Thank you for any help on this. I am so curious to hear some sound coming from the board.
all the best! 
-paul

marcel-licence, 2022-08-30:
Hi,
the loop area of the Demo sound might be sound a bit odd. I am not sure if there is a problem or just the odd sound.

To playback sounds from LittleFS you can use the drumcomputer project which is capable of a direct playback. Using the sampler project you need to load the sample into the PSRAM using the code like this:
```
    PatchManager_SetDestination(0, 1);
    Sampler_LoadPatchFile("/samples/pet_bottle.wav");
```
The settings for the sample will fall back to default if no binary file can be found containing the information of a loop point etc. 
The sound will be assigned to the next free channel. In addition to that channel 0 has one sound assigned for each key.

In https://github.com/marcel-licence/esp32_midi_sampler/blob/main/z_config.ino you will find a lot of connected functions which are used to control the sampler. There was a discussion in the past with some details how to set up a sound without any midi controller.

Best regards,
Marcel

baesek, 2022-09-07:
Hi Marcel, 
thank you ! now I understood the difference between the drumcomputer and the sampler project. Do you know the difference in latency the midi_sampler has in relation to the drumcomputer as it plays back the samples from PSRAM instead of the LittleFS? 
Again Just for understanding the two concepts better:
The drumcomputer can can have a maximum of 4MB (around 45s of 16bit 44100 mono audio) split into maximum of 12 samples readily available for low latency playback directly from the littleFS ?

The midi sampler sampler loads a maximum of also 4MB from either sd card or the littleFs to the PSRAM for playback.
The advantage is, that it can have a collection of samples on the sdcard which then can be loaded into the PSRAM when needed. I suppose this is not happening fast anough to trigger sounds from sd card directly without noticable latency.

I am just rethinking if the littleFs can really hold 4MB or is it 4MB - size of program? I am seeing the Partition scheme Setting of the Arduino Ide: Default 4MB with Spiffs (1.2MB App / 1.5MB SPiffs) 
Does this mean maximum flash memory for samples is also 1.5MB (15.8 seconds) ?

marcel-licence, 2022-09-11:
SPIFFS does contain a kind of file system. So you would have a bit less memory of raw data. In case of storing 16bit 44100 samples it would end up by approx 15 seconds. 
Regarding the SD card, yes streaming is very limited. In addition to that you would get problem when reading multiple file at once. I worked on this topic when doing the mellotron like sampler project: https://youtu.be/U5Q8chfMglE
From SPIFFS you can read multiple files at once and finally PSRAM behave more like RAM. Its much faster.