marcel-licence/esp32_midi_sampler #69: Sound distorted when VT100 is enabled

baesek, 2022-08-29:
Hi, 
I discovered another Problem with the Esp32 Audio Kit V2.2 2957 ES8388 Board. When I 
`#define VT100_ENABLED  `
The sound playback is very distorted. 
Is this normal ? What do you select in the Arduino Ide 's Tools menu? 
I have the following settings there:
Board: "ESP32 Dev Module"
Upload Speed: "921600"
CPU Frequency: "240MHz (Wifi/BT)
Flash Frequency: "80MHz"
Flash Mode: "QIO"
Flash Size: "4MB (32Mb)"
Partition Scheme: "Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)"
Core Debug Level: "Verbose"
PSRAM: "ENABLED"
Arduino Runs On: "Core 1"
Events Run On: "Core 1"
Port: "/dev/cu.usbserial-0001

Is this right?
regards,
paul

marcel-licence, 2022-08-30:
Hi Paul,

the distortion might be caused by the Core Debug Level "Verbose". The core will/might be busy with creating a lot of debug messages. 
The rest looks okay to me.

Best regards,
Marcel

baesek, 2022-08-30:
Hi,
I set the Core Debug Level to "None" but the sound is still super distorted when the VT100 output is enabled.

marcel-licence, 2022-09-24:
Hi,

I can observe the same issue when using the ESP32 board library version 2.0.2. Using 1.0.6 shows no issues. I had a lot of issues moving to the newer version. 
It sounds like that the Serial output call blocks the audio for each "frame" / textblock. 
I will try using the latest (2.0.5).

marcel-licence, 2023-04-21:
Hi @baesek , do you still observe the issue?

closed 2024-01-08 by marcel-licence