marcel-licence/esp32_midi_sampler #86: ESP32-LyraT Noisy sound and more issues

baesek, 2022-09-26:
Hi,
I am trying to use the esp32_midi_sampler with the ESP32-LyraT v4.3 Board at the moment and I get a really noisy sound which is covered by a beep tone in the frequency of the BLINK_LED_PIN.
So I commented out the blink process code which removed the repeating beep tone, but what stays is a steady noise floor and really silent sample playback.
I also realized that there is no Sound at all when i comment out pinMode(BLINK_LED_PIN, OUTPUT)
I outputted the value of the BLINK_LED_PIN and got 19 in the serial monitor.

Can you help me to get it work? I got it working with the drum_sampler_project but the midi_sampler is not working yet.

I cannot find where BLINK_LED_PIN is defined i think it is defined in the ml_synth lib

best,
paul

marcel-licence, 2022-09-26:
Hi @baesek,
I could imagine that there might be a pin conflict. I do not have the LyraT board. In some cases there is an amplifier placed on the PCB which can be set on/off. 

Referring to https://espressif-docs.readthedocs-hosted.com/projects/esp-adf/en/latest/design-guide/dev-boards/board-esp32-lyrat-v4.3.html you should replace IO19 by IO22 to get a blinking LED. IO19 should be a headphone detection input.

IO21 might be important to turn on/off the amplifier.

I would like to create a new file for LyraT and place it here: https://github.com/marcel-licence/ML_SynthTools/tree/main/src/boards to have all the pins defined. That could make it easier to select the setting with just one define in the config.h

Hope this helps

marcel-licence, 2023-04-21:
No update

closed 2023-04-21 by marcel-licence