marcel-licence/esp32_midi_sampler #97: What pins for connecting SD Card Module to ESP32 Devkit V1 DOIT?

satriamahen, 2023-11-14:
I got everything working as I intended, just missing the SD Card functionality. Because I don't know where to connect the SD Card module to the ESP32 Devkit V1 DOIT. Can anybody help me? Thanks in advance!

satriamahen, 2023-11-14:
[image]
Will this connection work?

marcel-licence, 2023-11-16:
Using SPI you would be flexible with the used pins. It can be mapped by the ESP32. If you need a fast connection you should use SD_MMC. It can use 1 or 4 data lines. In case you use SD_MMC pull-up resistors may be required.

Did you run some example code to test the sd-card?

closed 2024-01-08 by marcel-licence