marcel-licence/esp32_midi_sampler #101: The sound stutters when changing any CC message

musikkfeil, 2025-01-18:
My board is esp32 audiokit v2.2 A417. 
MIDI is connected using H11L1 according to your scheme.  I use - "AUTO_LOAD_PATCHES_FROM_SD_MMC"
It seems that all functions work (loading files, playing notes and changing CC via serial midi). Display also works. But..
I tried different boards 'esp32' Version 1.0.6 to 2.0.17. but on all of them the same problem - if the sound is playing and at the same time you change the CC message, the sound stutters during the change. Is it just me?

marcel-licence, 2025-01-18:
It might be possible that the text output on the serial monitor takes too much time and could create the stutter. Do you know if there is any output? This project is more or less deprecated and should be replaced by [the ml_midi_sampler](https://github.com/marcel-licence/ml_synth_sampler_example/tree/main)
Do you experience the stutter on all CCs or just certain CC messages?
Maybe an audio example might help for better identification of the problem.

musikkfeil, 2025-01-18:
on all CCs !
[](https://soundcloud.com/malinin-device/1a1)
[](https://soundcloud.com/malinin-device/2a1)
The first part of the file is just sound, the second is me trying to send СС message. In this case I change decay and sustain.
I tried ml_sampler - it doesn't have such problem. But it doesn't have many functions that this sampler has. And also the sound is worse on some notes. (

musikkfeil, 2025-01-18:
https://soundcloud.com/malinin-device/1a1
https://soundcloud.com/malinin-device/2a1

marcel-licence, 2025-01-18:
Thank you for the audio demo.
It seems it is overloaded with some task and not be able to process the audio fast enough. I never experienced a problem like this.

Which features do you miss on the other sampler? I should plan to port them.
A worse sound? Do you have an example for this too?

musikkfeil, 2025-01-18:
about overload - where do you think this could cause such a problem - where do I need to dig?

about the sampler
Unfortunately I didn't find such important functions for me as:
recording and saving new samples
loading samples on the fly
loop point control (end, start, length)
full control on adsr

sound. the first part is esp32_sampler the second is ml_sampler. 
example with the same wav file. 
https://soundcloud.com/malinin-device/3a1

esp32_sampler is cleaner on all notes - ml_sampler only on the first one, on the others there is some kind of hiss noise.

musikkfeil, 2025-01-18:
So. I just downloaded a clean zip esp32 sampler again. And I don't see this overload. So it was me who did it somehow. I'll analyze what I did ) Sorry...

marcel-licence, 2025-01-18:
Regarding the sampler, I could check if I could merge missing functions and compare the sound core.
The hiss could be some aliasing. I made the code of the new sampler sound core completely open and this might be a bit simplified in comparison to the esp32 sampler.

Good to know that there is no issue with the playback.

closed 2025-01-18 by musikkfeil